FaradayRF/Faraday-Firmware #4: RF Layer 2 - Callsign Length

kb1lqd, 2016-10-09:
RF Layer 2 packet does not support a callsign longer than 6 bytes. This should be updated to 9 bytes to match the longest expected callsign and suffixes. All other layers and programs support or will support a 9 byte callsign.

Excluding non United States callsigns (which can be longer than 6) support for 9 achieves designators for location/operation type:

Normal: KB1LQD
Maritime mobile: KB1LQD/MM
Mobile in region 3: KB1LQD/3

[image]

jwnelson, 2017-03-06:
I'm a bit confused over the packet length definitions. First, TX_PACKET_LEN is defined to be 61, while RF_DATALINK_PACKET_PAYLOAD_LEN is 62. What is the distinction between the two defines, and why are they different? Considering the two CC430 append bytes and the 63 byte limit imposed by #5, shouldn't the TX packet length be 61 and the RX packet length be 63?

kb1lqd, 2017-03-13:
@jwnelson

## rf.h

```
/** @name Radio Transmit/Receive Layer 2 Packet Definitions
* 	@brief Basic low level packet definitions for Layer 2 (Datalink)
@{**/
#define TX_PACKET_LEN 61 /**< Transmit packet size */
#define RX_PACKET_LEN 61 /**< Receive packet size */
#define RX_PKT_HANDLE_APPEND_LEN 2 /**< Packet hardware receive append data byte size */
/** @}*/

/** @name RF Layer 2 FIFO Definitions
* 	@brief Layer 2 definitions that support the FIFO buffers
*
* 	@bug Why is RF_DATALINK_PACKET_PAYLOAD_LEN 62 bytes instead of 61? RX extra footer bytes?
* 	@bug All of this should be updated to be in the main definitions and not FIFO specific!
*
@{**/
#define RF_DATALINK_PACKET_PAYLOAD_LEN 62 /**< FIFO definition for packet length */
#define RF_DATALINK_PACKET_FIFO_COUNT 5 /**< FIFO definition for max number of packets to hold in a FIFO */
#define RF_DATALINK_PACKET_RX_FOOTER_LEN 2 /**< FIFO definition for receive footer byte count */
/** @}*/
```

`RF_DATALINK_PACKET_PAYLOAD_LEN` is used to create the size of the RF FIFO for both RX and TX.
```
volatile unsigned char rf_datalink_tx_fifo_buffer[RF_DATALINK_PACKET_PAYLOAD_LEN*RF_DATALINK_PACKET_FIFO_COUNT]; /**< Transmit FIFO buffer */
```

```
volatile unsigned char rf_datalink_rx_fifo_buffer[(RF_DATALINK_PACKET_PAYLOAD_LEN+RF_DATALINK_PACKET_RX_FOOTER_LEN)*RF_DATALINK_PACKET_FIFO_COUNT]; /**< Transmit FIFO buffer */
```

`TX_PACKET_LEN` references:
[image]

`RF_DATALINK_PACKET_PAYLOAD_LEN` references:

[image]

# Issue Case(s)

## rf.c: rf_get_next_tx_fifo()
```
void rf_get_next_tx_fifo(void){
	get_fifo(&rf_datalink_tx_fifo_state_machine, rf_datalink_tx_fifo_buffer, (unsigned char *)rf_tx_datalink_buffer);
	//Transmit packet from queue
	Transmit(&rf_tx_datalink_buffer,TX_PACKET_LEN);
}
```

The FIFO element `rf_datalink_tx_fifo_state_machine` contains data packets defined as length=63 but uses `TX_PACKET_LEN` which is length = 61.

The put() variant uses a function argument length variable:

```
void rf_tx_put_packet_buffer(unsigned char *packet_data_pointer, unsigned char length){
	put_fifo(&rf_datalink_tx_fifo_state_machine, &rf_datalink_tx_fifo_buffer, packet_data_pointer);
}
```

where the `put()` is passing into the arguement a length of 62 bytes:

```
void rf_rawpacket_tx(RF_DATALINK_PACKET_STRUCT *packet_data_pointer){
	__no_operation();
	rf_tx_put_packet_buffer((unsigned char *)packet_data_pointer, 62); //Chang length to #DEFINE
	__no_operation();
}
``` 

@jwnelson you are correct this is all sorts of inconsistent!

kb1lqd, 2017-03-13:
```
DEFINES:
TX_PACKET_LEN 61
RX_PACKET_LEN 61
RX_PKT_HANDLE_APPEND_LEN 2
RF_DATALINK_PACKET_PAYLOAD_LEN 62
RF_DATALINK_PACKET_RX_FOOTER_LEN 2

Variables:
tx_buffer[62];

FIFOS:
rf_datalink_tx_fifo_buffer[RF_DATALINK_PACKET_PAYLOAD_LEN*RF_DATALINK_PACKET_FIFO_COUNT] = 62*x bytes
rf_datalink_rx_fifo_buffer[(RF_DATALINK_PACKET_PAYLOAD_LEN+RF_DATALINK_PACKET_RX_FOOTER_LEN)*RF_DATALINK_PACKET_FIFO_COUNT] = 64*x bytes

Functions:

```

kb1lqd, 2017-03-13:
@jwnelson I created https://github.com/FaradayRF/Faraday-Firmware/issues/71 to further track this cleaning.