FaradayRF/Faraday-Firmware #15: Telemetry Readings Have "Tick"

kb1lqc, 2016-10-18:
I noticed that the ADC readings seemed to have a "tick" to them with my new telemetry application. Then I checked the Rev B Faraday with the Balloon Launch software (8/2016) using the same firmware and there it was, still ticking:

**ADC Channel**
[image]

**GPS Altitude**
[image]

This indicates it's not my new parser, it's in the firmware. Turns out it's also on the GPS data so this seems to clearly be in the firmware land and not just with the ADC code. What did you recently change in firmware to possibly cause this?

kb1lqc, 2016-10-18:
That's weird, if I plot Altitude (Radio Unit) yellow and AP Altitude (Local Unit) green which in this case is just my own unit then the local unit data doesn't appear to have the "ticks". Since this should be just my locally connected unit I would expect both of these to be exactly the same, the Balloon Software I tested this on never did this...

[image]

kb1lqd, 2016-10-18:
@kb1lqc Can you post a snippet/file of just relevant data? Are the ticks actually IN the raw data from telemetry/SQLite?

kb1lqc, 2016-10-18:
Actual Board Temp: Column AF (Labaled ADC8)
VCC Battery: Column AE (Labeled Board Temp)

[telemtest.zip](https://github.com/FaradayRF/Faraday-Firmware/files/535553/telemtest.zip)

kb1lqc, 2017-02-06:
OK I'm closing this. No root cause, a byproduct of moving quick over many different items and starting our manufacturing run. Will not be acceptable in the future but for now need to accept it. I assume this was fixed with something @kb1lqd changed in the handling of telemetry. Likely while addressing the VCC and board temperature firmware bugs

Below is the current plot of most ADC channels using "Faraday_D1_Release.txt" firmware file and the current master software (https://github.com/FaradayRF/Faraday-Software/commit/4c12933766e6f7c8891d682348f39f542e878552)

[image]

All seems well.

closed 2017-02-06 by kb1lqc