FaradayRF/Faraday-Firmware #54: Misspelled structure member

k5okc, 2016-12-22:
"source_indetifier" should be "source_identifier"

In directory RF_Network_Stack

rf.c:382:		rf_datalink_packet_tx_struct.**source_indetifier** = src_callsign_id;
rf.c:436:	rf_datalink_packet_rx_struct.**source_indetifier** = packet[RF_DATALINK_PKT_SOURCE_CALLSIGN_ID_LOC];
rf.h:75:	unsigned char **source_indetifier**; /**< source callsign ID number */

kb1lqd, 2016-12-23:
Confirmed misspelled structure member on lines 382 and 436 of RF.c:

[image]

Refactored name to "source_identifier":

[image]

Recompiled successfully.

Reprogrammed two REV D1 units using the Bootstrap Loader (No JTAG):

[image]

Succesfully ran several RF tutorial programs including a the simple RF text message tutorials which shows that RF.c (Layer 2 of the RF network stack) is still working as intended.

[image]

I'm considering the update a success and bug to be closed. Thanks @k5okc !

closed 2016-12-23 by kb1lqd

kb1lqd, 2016-12-23:
I Realized that the screenshot above didn't show the source callsign and identifier working. It does, not sure why it didn't come through on the first screenshot:

[image]