FaradayRF/Faraday-Firmware #67: Indicate RF Transmission by LED

kb1lqc, 2017-02-22:
Per comments via email it would be good to at least default provide indication of transmission with an LED blink. We could dedicate an entire LED by default to transmission. This would help the user know if their unit is transmitting and when it is. Helping to alleviate some question about whether a setup is incorrect with software or if the radio is not transmitting.

It would be advantageous to be able to turn this functionality on/off via software configuration therefore allowing it to be used when desired but not interrupt LED use when undesired.

kb1lqd, 2017-02-22:
Yep, this is pretty easy. I can throw it in the general bitmask for now in a device configuration to enable/disable. It would remove the users ability to work with the affected LED when enabled.

This would be a firmware update. Shouldn't be hard at all though, I actually use to have it in hacked form...

kb1lqd, 2017-06-24:
Starting work on a simple version of this using the boot config bitmask. Just for you two @kb1lqc  and @el-iso !

https://github.com/kb1lqd/Faraday-Firmware/tree/issue67

kb1lqd, 2017-06-24:
OK it is working and setting `CONFIGBOOTBITMASK` in the device configuration (`faraday_config.ini`) `BIT1` HIGH will enable the operation. Note that unless you want to erase your configuration always keep BIT 0 HIGH. This means that to enable the LED set `CONFIGBOOTBITMASK = 3` to have binary `00000011`.

kb1lqd, 2017-07-07:
https://github.com/FaradayRF/Faraday-Firmware/pull/76 closes this issue.

closed 2017-07-07 by kb1lqd