FaradayRF/Faraday-Firmware #69: RF not receiving due to callsign settings

kb1lqd, 2017-02-28:
## Summary
Working with an owner of a Faraday REV D1 it has been found that certain callsign or other settings may cause the RF to cease reception. Configurations known to produce proper operation allow transmissions but the units do not recieve data into Proxy. Known good units have reproduced the problems thus indicating a likely firmware bug.

## Problem Explanation
Owner K6YAP reported problems receiving signals from Faraday. I have reproduced the problems using his exact callsign in an RF test setup with known good units. Telemetry broadcasts RF layer 2 MAC packets addressed to "CQCQCQ" for general broadcasting but debugging has shown:

Unit 1:
Callsign = K6YAP
NodeID = 0

Unit 2:
Callsign = K6YAP
NodeID = 1

* Unit TX: Destination = "CQCQCQ"
* Unit RX: Destination = "CQCQC0xD0" where the 0xD0 is a non-ASCII hex value and likely a random memory byte during packet creation (may not be the same on all devices).

When receiving packets the incorrect receive destination callsign shown above causes the check to fail and packet to get dropped from BOTH the callsign check and CQCQCQ check. This occurred on both units receiving each-others telemetry and is the most probable cause of receive path failing for K6YAP.

### Environment
#### Software 
* Firmware = Master
* Software = Master

#### Hardware
* Unit SN's (KB1LQD's): SN12, SN13
* Operating System = Windows 10

# Supporting Information

### Receiver Layer 2 MAC "CQCQCQ" Destination Check Fail

[image]

kb1lqc, 2017-02-28:
Nice work @kb1lqd !

kb1lqd, 2017-02-28:
Created firmware branch https://github.com/FaradayRF/Faraday-Firmware/commit/c1489a4c5d4c69d2822510d482b463ef437bceff

kb1lqd, 2017-02-28:
@kb1lqc OK I'm not sure how you want to review this. Did you want to run it or just review the single variable change and approve?

If I make a pull request I'm going to have to update the firmware variable. I'd like K6YAP to attempt to program using the BSL loader but knowing he's on a Mac I'm not sure if it will work out of the box. I guess that may be the next project.

kb1lqd, 2017-02-28:
@kb1lqc and Reilly, can you try using the BSL to program this into your units and verify working?

[Faraday_D1_Release_Bug_69.txt](https://github.com/FaradayRF/Faraday-Firmware/files/806165/Faraday_D1_Release_Bug_69.txt)

I have not tried the BSL on anything but windows. It uses an .exe so maybe it'll work under Wine.

https://github.com/FaradayRF/Faraday-Software/tree/master/Firmware_Bootstrap_Loader

Getting BSL cross platform may be the next step as this needs to be pushed to ALL Faraday owners in an official release. We also should point out this bug to anyone got/getting units with the currently (initial release) firmware.

kb1lqd, 2017-03-01:
Uploaded below is known good BUG 59 bootstraploading ZIP.

[BUG_69_Firmware_Bootstrap_Loader.zip](https://github.com/FaradayRF/Faraday-Firmware/files/809824/BUG_69_Firmware_Bootstrap_Loader.zip)

binaryfox0, 2017-03-02:
I've verified that the firmware from the ZIP above fixes my issues.

reillyeon, 2017-03-02:
I've verified that firmware from the ZIP file above fixes this issue.

kb1lqd, 2017-03-12:
Fixed in #70

closed 2017-03-12 by kb1lqd