FaradayRF/Faraday-Firmware #72: GPIO State Sends Down RF IO Values

kb1lqc, 2017-04-11:
Per the image below of the IO telemetry sent down over RF, GPIO state contains three values from RF IO.
[image]

GPIO0, GPIO1, and GPIO3 are incorrectly set as P4.7, P4.6, and P4.5 respectively when compared below:
[image]

The correct values for the GPIO signals in question include:
[image]

This Telemetry Error was caused by [Hardware Issue #6](https://github.com/FaradayRF/FaradayRF-Hardware/issues/6)

This will affect critical applications such as the RHIT HAB which they should perform a firmware update to fix this ASAP.

kb1lqc, 2017-04-19:
OK so most of this is working. It appears as though GPIO0, GPIO1, and GPIO2 are not working as expected. They are always  (1,1,0) respectively.

Per the documentation these should be the following P3.x pins. However as shown above it doesn't appear to be send down in GPIOSTATE or IOSTATE.
[image]

kb1lqd, 2017-04-19:
I forced all GPIO telemetry bits to be 0 (note that the GPIO bitshifting is NOT present in master):

[image]

I've confirmed the 2 stuck bits in raw unpacked telemetry:

[image]

This would show up in simple UI as GPIO-6 and GPIO-7:

[image]

kb1lqd, 2017-04-19:
@kb1lqc can you explain what this is doing in detail? I'm wondering if there is a parsing / shift error here.

https://github.com/kb1lqc/Faraday-Software/blob/simple_web_ui/Applications/SimpleUI/static/simpleui.js#L50

kb1lqd, 2017-04-19:
OK I found that the GPIO state was always 48 which is '0b110000' which is also HEX 0x30 or ASCII '0'

This led me to realize HDOP might have something to do with this because in the no-GPS case it is set to an array of ASCII `0`'s.

I found that AFTER I set GPIO State in telemetry framer on the MSP430 that I also copy in `telem_packet_3_struct) which holds a GPIO state... As shown below I copy IO and RF state into this struct but not GPIO state which remains ASCII '0':

[image]

[image]

[image]

I should even need the struct `telem` and instead directly use `telem_packet_3_struct`:

removing the unneeded `telem` variable and directly editing the telemetry frame struct:

[image]

I see correct GPIO output!:

[image]

kb1lqd, 2017-04-19:
@kb1lqc 

Below is the firmware that fixes:
* Updated telemetry packing function to actually pack correct GPIO State which wasn't occurring prior.

Reference: https://github.com/FaradayRF/Faraday-Firmware/tree/swissue153
Firmware File: https://github.com/FaradayRF/Faraday-Firmware/blob/swissue153/Debug/Faraday_D1_Release.txt

Please tryout out and respond.

kb1lqd, 2017-05-07:
Closing since bug was fixed in merges.

closed 2017-05-07 by kb1lqd