FaradayRF/Faraday-Firmware #75: Create Simple RF Packet Application

kb1lqd, 2017-05-13:
Create a simple packet application that allows all data received on a UART port to be forwarded to RF and visa-versa. This will be simular to the command modules "rf packet forwarding" command except:

- Be specific to a port
- Provide FIFO's

This will help facilitate creating RF applications in the future and also begin providing a means to measure throughput and other bottle necks within Faraday.

kb1lqd, 2017-05-14:
I verified the bare-bones FIFO functions needed to tie a new program into the UART FIFO architecture:

# rf_packet.c

- void app_init_app_rf_packet(void);
  - Initialization of FIFO state machine and FIFO
- void app_rf_packet_uart_rx_put(unsigned char *data_pointer, unsigned char length);
  - PUT() function for FIFO that is used as a function pointer address location from the UART port "OPEN" array. This places data in to the FIFO.
- void app_rf_packet_housekeeping(void);
  - Application housekeeping routine used to perform functions on each general housekeeping pass

# main.c

## initialize FIFO's

`app_init_app_rf_packet();`

## Open UART port to PUT() Function

`uart_service_open(1, app_rf_packet_uart_rx_put); //UART RF Packet Link`

# housekeeping.c

Add application housekeeping function to raw housekeeping function `housekeeping_check_raw()`.

`app_rf_packet_housekeeping();`

[RF_Packet.zip](https://github.com/FaradayRF/Faraday-Firmware/files/999012/RF_Packet.zip)

kb1lqd, 2017-05-14:
I've confirmed RF output using a simple hook into the above source code: 901b9fc097e8a18c81fbdc4e4a89a1709a001da2

kb1lqd, 2017-05-14:
I fixed a bug that cause incorrect packets to transmit and corrupt reception. I also added an RF receive ability and proved it working!

d522d417d344ba325bc79884f39f7733fc686f45

kb1lqd, 2017-05-14:
I created a quick demo video of the current progress: https://youtu.be/QbDp6pvj2n0

kb1lqd, 2017-05-20:
I've tested some of the buffers and found that RF Packet app, UART and RF buffers seem to NOT be overflowing when sending fast data. I've set up the delays to cause them to overflow to verify.

I also found that when running fast single transmissions the "sounds" of stuttering are reduced on the SDR if I remove python print statements...

kb1lqd, 2017-05-20:
Here's my priority plan:

- developer tool to measure RF throughput from faraday
- upgrade post command to accept batch data
- fix any quickly overflow or other bottlenecks
- assuming decent speed attempt digital audio stream
  - https://www.opus-codec.org/

kb1lqd, 2017-05-21:
Oh snap! I made a simple tool to measure throughput and got ~17kbps solid when I wasn't recording video. About 15kbps when I was.
6711f0023f0472f34deca41e37bcc33bc6d37d45

https://www.youtube.com/watch?v=lLm2hJKPsNE&feature=youtu.be

kb1lqc, 2017-05-24:
Very cool! Thanks @kb1lqd

kb1lqd, 2017-06-01:
OK so I've got a base to work with now where Faraday is transmitting and receiving frames pretty well. Do note that I need the following software bug fixes fixed in MASTER eventually:

https://github.com/FaradayRF/Faraday-Software/issues/202

Un-Reported issue: Proxy seems to NOT accept multiple packets as an array as advertised. @kb1lqc @reillyeon If they have any thoughts on this.

Currently the RFDataport program accepts properly sized data packets on localhost port 8009 and sends those data packets over RF to "CQCQCQ" which is the "broadcast" callsign for all units. I believe making a proxy program that accepts all data sizes and fragments as needed is ideal. Note that in THIS push we should do as much processing on a computer as possible and leave Faraday simply as the "modem".

**Actions**
* Add 'Control Frame" as a packet type: This will be used to set RFDataport remote callsign to transmit to and possibly other settings. Also, Faraday should respond to a request for information of who it is transmitting too in this frame.
  * This will need some firmware updating as well to support "querying" for program state
* RFDataport Server should accept data of ANY length and the server should fragment as needed
* N8VNR is helping creating an OPUS audio stream to input into RFDataport to achieve digital audio transmission. This will supply OPUS encoded frames on localhost that can be directed to "RFDataport".

**Possible Test Instances**
Faraday is by no means optimized and there are likely settings and features for the CC430 / RF stack that would improve the error rate. After formalizing the RFDataport framing structure and fragmentation support I should write a quick end-to-end test platform to generate test data, record received data, calculate drop/error rate. This will allow us to play with internal register settings like CC430 SYNC headers, AGC, etc... Also, this will help us find any bottlenecks or bugs in the FIFO's.

kb1lqd, 2017-06-02:
I added:

* RFDataport Server should accept data of ANY length and the server should fragment as needed

https://github.com/kb1lqd/Faraday-Software/commit/5425f5d982210c5de2495b6718fb9f7f42cbe044

kb1lqd, 2017-06-05:
I wrote a basic text message program and found some bugs in the software! 

https://www.youtube.com/watch?v=enWD4N6ufT8&feature=youtu.be

Commit: https://github.com/kb1lqd/Faraday-Software/commit/1e3e55642c47e0bddf233ff96f7a59a4a0618990

kb1lqd, 2017-06-26:
Created updated merge branch: https://github.com/kb1lqd/Faraday-Firmware/tree/rfdatajune2017

kb1lqd, 2017-06-26:
@kb1lqc 

Current Status:
* Firmware rebased branch to master and merged known working branch of firmware that brings in the RFDataport functionality to the Faraday unit. 

These are the items needed to get RF Data application running end to end:
* Proxy must connect to multiple local units (to transfer data between the same computer
* Get RFDataport_server.py running with updated proxy as of June 2017 (pip install proxy version)
  * https://github.com/kb1lqd/Faraday-Software/tree/fw_issue_75/Applications/RFDataport
* Verify basic data transfer between units

### Testing with POSTMAN
This is a simple way to test transmit and receive using the RFDataPort Server python application and POSTMAN (instead of specific TX and RX scripts).

**Sending a data packet**
[image]

**Receiving a data packet**
[image]

kb1lqd, 2017-07-07:
https://github.com/FaradayRF/Faraday-Firmware/pull/78 closes this issue.

closed 2017-07-07 by kb1lqd