FaradayRF/Faraday-Firmware #81: Using the CP2120 For I2C

kb1lqd, 2017-10-05:
A carry over from #80 SPI testing but now focusing on the I2C bus.

## General Notes

### I2C - Reading From a Device

To read from a device over I2C (and the CP2120) you need to first write to the bus. Specifically you need to write to the the device write address followed by the register of information you want to retrieve later. In the case of the CP2120 setup it is common to get an I2C response of `NAK` and a retry is required.

``` c
	C2120_Write_Reg_I2C(BMP180_ADDRESS_WRITE, 0xD0);
	testspi = 0x00;
	while(!(testspi==0xF0)){
			__delay_cycles(SPI_BYTE_CYCLES);
			testspi = C2120_Get_I2CSTAT();
			//If NAK received try again
			if(testspi==0xF1){
				C2120_Write_Reg_I2C(BMP180_ADDRESS_WRITE, 0xD0);
			}
		}
```

When successful the loop will end and the device on the bus is ready to be "Read" from. This is achieved by using the `C2120_Read_Reg_I2C(BMP180_ADDRESS_READ, 1);` to read 1 byte of data from the device READ address (the register data to be returned has already been determined in the prior WRITE command to the device).

kb1lqc, 2017-10-05:
Oh nice find on the `NAK` response!

kb1lqd, 2017-11-06:
Closing due to scope reduction, not adding I2C sensors and focusing on core telemetry and analog to then move onto other Faraday uses such as TUN network adapter and RPI integration.

closed 2017-11-06 by kb1lqd